coloradocube/balloonsat #28: Main flight loop

ivogeorg, 2021-11-05:
# Description

Main flight loop has to be stable and easy to recover. Preferably, multi-threaded (RPi 4 has a quad-core processor). 

# Knowledge

1. Python [Thread-base parallelism](https://docs.python.org/3/library/threading.html). Due to Global Interpreter Lock, only one thread can execute at a time, still suitable for multiple I/O-bound tasks. For better use of multi-core processing resources, `multiprocessing` or `concurrent.futures.ProcessPoolExecutor` should be used.

# Deliverables

# Time estimate

NotfatPrince, 2021-11-13:
[image]

NotfatPrince, 2021-11-13:
[image]
[image]
[image]

ivogeorg, 2021-11-14:
Have to design the WiFi streaming for the demo @NotfatPrince @Christopher-Dauchter

ivogeorg, 2021-11-15:
## Update

[image]

ivogeorg, 2021-11-15:
## Update summary

1. System Monitor part of Flight Executive.
2. Preflight Check's acceptance rules derived from telemetry rules.
3. Split log into Telemetry Log, Camera Log, (Iridium/Comms) Connection Log, Executive Log.
4. Added Rules to:
   1. Camera managing storage.
   2. Camera particular moments of interest.
   3. Telemetry rules in System Monitor.
5. Iridium sends 3 types of messages:
   1. Periodic (automatic) updates.
   2. Responses to data requests from ground.
   3. Critical alerts sent by the System Monitor.
6. Connection status removed from Telemetry log.

ivogeorg, 2021-11-15:
## Update

[image]

ivogeorg, 2021-11-15:
## Update summary

1. The Flight Executive is the main process.
2. The Flight Executive has stages:
   1. Startup on Power-On.
   2. Pre-flight check & acceptance tests.
   3. Mission & monitoring.
   4. Post-flight operations.
3. The Acceptance Rules are:
   1. Part of the telemetry rules.
   2. Applied in Startup on Power-On and Pre-flight Check & Acceptance Tests.
   3. Applied in stages:
      1. Critical zone: power only.
      2. Acceptance zone: pre-flight check.
      3. Mission alerts: monitoring of mission logs and reporting critical alerts to ground.

ivogeorg, 2021-11-23:
@PaulAJanes This should give some more context of what you are doing...

The Flight Executive (FE) is:
1. The top process in the https://github.com/MSUSAT/balloonsat directory, with `telemetry`, `camera`, and `rockblock` being packages it imports. The structure should be like this (with any other necessary sub-packages):
   ```
   balloonsat/
   |--executive.py
   |--telemetry/
      |---__init__.py
      |---telemetry.py
   |--camera/
      |---__init__.py
      |---camera.py
   |--rockblock/
      |---__init__.py
      |---rockblock.py
   ```
2. Is a state machine. State machines are much easier to write and maintain than non-systematic code. The FE states are the 4 main stages mentioned before, along with any other identifiable states that are important to segregate. The states are an `enum` class. This is a tentative list of the states: `POWER_ON, STARTUP, PREFLIGHT_CHECK, MISSION, POSTFLIGHT`. The transitions between states are strictly identifiable functions.

ivogeorg, 2021-11-25:
Major FS design points:
1. The FE and the 3 apps are processes, to potentially each run on a separate core.
2. Each of them may be multi-threaded, if necessary.
3. They communicate with message queues, callbacks, and/or a pub-sub bus. _Which one is best for cross-core comms?_

ivogeorg, 2021-11-30:
# FE cannot fail

1. The FE **cannot fail**. This means that it should:
   1. Be minimally complex in its core. 
   2. Hand off more complex behavior to threads or sub-processes.
   3. Should have a catch-all for exceptions with a _simple recovery strategy_.
   4. Should not be involved in functionality that can result it deadlock.
   5. Should communicate with other processes in a non-committal manner, preferably pub-sub.
   6. Should not write to the filesystem.
2. _Should it be a [deamon process](https://www.python.org/dev/peps/pep-3143/)?_ (Related, from the Python Cookbook: [Forking a daemon process on Unix](https://learning.oreilly.com/library/view/python-cookbook/0596001673/ch06s08.html#pythoncook-CHP-6-SECT-8.1), Google [search results](https://www.google.com/search?q=python+daemonize+process&rlz=1C1GCEB_enUS972US972&oq=python+process+daemonizer&aqs=chrome.1.69i57j0i22i30j69i64.8146j0j7&sourceid=chrome&ie=UTF-8))

ivogeorg, 2022-01-26:
# FE command control

There should be a way to **start** and **stop** the FE without Ctrl+c. This is potentially just as damaging as hard shutdown of the RPi (from the perspective of the `fexecutive.py` process, it's the same thing).

Potential solutions:
1. (easy) For short-term testing of the `fexecutive.py` code, there can be a time interval argument for the flight loop to stop when the time since the start has elapsed.
2. (a bit more involved) This can be extended to shutting down after some interval after landing.  
3. (difficult) Full-on daemon server.