coloradocube/balloonsat #47: FE v0.0.4 - Monitor if RPi is being throttled

ivogeorg, 2021-12-06:


dperry29, 2021-12-06:
Command to check if Rpi is throttled due to low voltage:
  vcgencmd get_throttled
  # Pi will return: "throttled=0x50005"

<html><body>

Bit | Hex Value | Meaning
-- | -- | --
0 | 1 | Under-voltage detected
1 | 2 | ARM frequency has been caped
2 | 4 | Currently throttled
3 | 8 | Soft temperature limit is active
16 | 1000 | Under-voltage has occurred
17 | 2000 | ARM frequency capping has occurred
18 | 4000 | Throttling has occurred
19 | 8000 | Soft temperature limit has occurred

</body>
</html>

ivogeorg, 2021-12-07:
Interesting. These commands are polling the Broadcom VideoCore GPU!

ivogeorg, 2021-12-07:
# Knowledge

1. https://pypi.org/project/vcgencmd/
2. https://elinux.org/RPI_vcgencmd_usage