coloradocube/balloonsat #50: FE v0.1.2 - Read and log GPS receiver

ivogeorg, 2021-12-06:


NotfatPrince, 2021-12-08:
Get the latest version of wiringpi
```wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb
gpio -v
```
Install python libraries
```sudo apt-get update
sudo pip install RPi.GPIO
sudo apt-get install python-serial
sudo apt-get install gpsd-tools  gpsd-clients 
sudo -H pip3 install gps3
```
Download demo codes 
```mkdir ~/Documents/MAX-XXX_GNSS_HAT_Code
cd ~/Documents/MAX-XXX_GNSS_HAT_Code/
wget https://www.waveshare.com/w/upload/0/0f/MAX-XXX_GNSS_HAT_Code.zip
unzip MAX-XXX_GNSS_HAT_Code.zip
```
Files should be in /home/pi/Documents/ MAX-XXX_GNSS_HAT_Code/RaspberryPi/python/coordinate_converter

ivogeorg, 2022-01-28:
# Knowledge

1. Adafruit [`gpsd` guide](https://learn.adafruit.com/adafruit-ultimate-gps-hat-for-raspberry-pi/use-gpsd).
2. An oldish [writeup](https://area-51.blog/2012/06/18/getting-gps-to-work-on-a-raspberry-pi/).
3. Slightly more recent [forums](https://forums.adafruit.com/viewtopic.php?f=50&t=70133&start=60#p359668).