coloradocube/balloonsat #59: FE v0.1.4 - Outdoor test of GPS

ivogeorg, 2021-12-06:


NotfatPrince, 2021-12-08:
Get the latest version of wiringpi
```wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb
gpio -v
```
Install python libraries
```sudo apt-get update
sudo pip install RPi.GPIO
sudo apt-get install python-serial
sudo apt-get install gpsd-tools  gpsd-clients 
sudo -H pip3 install gps3
```
Download demo codes 
````mkdir ~/Documents/MAX-XXX_GNSS_HAT_Code
cd ~/Documents/MAX-XXX_GNSS_HAT_Code/
wget https://www.waveshare.com/w/upload/0/0f/MAX-XXX_GNSS_HAT_Code.zip
unzip MAX-XXX_GNSS_HAT_Code.zip
```
Files should be in /home/pi/Documents/ MAX-XXX_GNSS_HAT_Code/RaspberryPi/python/coordinate_converter

ivogeorg, 2022-01-23:
RPi can read the GPS signal in [WGS48](https://gisgeography.com/wgs84-world-geodetic-system/) (lon, lat, alt) outdoors with an external antenna. Problems with the code will be solved in #50. Closing.

closed 2022-01-23 by ivogeorg